SmartGridready/SGrSpecifications #120: REQ - XSD Naming Conventions

ergozeller, 2023-08-25:
## User Voice
We should have clear naming conventions for the SGr XSD

## Suggested Solution
 ### Naming
 The following naming are applied:
 - elements, attributes, and types use camelCase for naming
 - no abbreviations
 - same things have same names
 - use meaningfull names that can be understood
 - if the element defines a value with units the name should have the unit as suffix (example: timeToLiveMs for milliseconds)
 
 ### Elements vs Attributes
As a rule of thumb: Relevant data should use elements, whereas meta data should be put in attributes
 If in doubt wheter to use elements or attributes ask yourself if you loose significant information when stripping the XML of all attributes. If yes, use an element, if no an attribute is fine.
 
 ### Namespace
 Use the SGr namespace as default namespace in the XMLs. 

## Acceptance Criteria
List the criterias that need to be met to declare this feature as implemented. This section must be complete and agreed upon before starting implementation.
- [ ] Naming conventions are documented
- [ ] XSD and all XMLs comply to the naming conventions

ergozeller, 2023-08-25:
Hi @ergo-furrer @ArndGerns 
Ist obiger Vorschlag für die Naming Conventions OK? Fehlt noch was? Bitte um euren Input...
S.

ArndGerns, 2023-08-25:
Ist `no abbreviations` nicht etwas hart? Du hast in timeToLiveMs auch eine Abkürzung `Ms`.

Und was ist mit den Typen? Wollen wir die mit Grossbuchstaben beginnen und Nicht-Typen klein?

Kleinigkeit: Tipp-Fehler `If in doubt wheter` fehlt ein `h`.

ergozeller, 2023-08-25:
> Ist `no abbreviations` nicht etwas hart? Du hast in timeToLiveMs auch eine Abkürzung `Ms`.

Hm ja, aber ausschreiben macht Sinn weil verständlicher. OK wenn wir "try to avoid abbreviations" definieren? So sind wir weicher...

> Und was ist mit den Typen? Wollen wir die mit Grossbuchstaben beginnen und Nicht-Typen klein?

Auch OK. @ergo-furrer was meinst Du dazu?

> Kleinigkeit: Tipp-Fehler `If in doubt wheter` fehlt ein `h`.

Korrigiert :)

> Sollen wir auch Formatierung definieren? (Blanks, Tabs, Einrücken wann / wieviel, ..., Umbrüche)?

Ja, Vorschläge?

ArndGerns, 2023-08-25:
`"try to avoid abbreviations"` ist besser.

Formatierung: **nur eine Diskussionsgrundlage**

- keine Tabs
- Einrücken um jeweils 2 (oder 4? wird aber vielleicht zu breit?)
- Zeilen möglichst nicht länger als 120 Zeichen
- Nur immer ein <element> <sequence>, ... in einer Zeile
- Wenn ein Tag über mehrere Zeilen geht (z.B. Beschreibung) Ende-Tag auf eigene Zeile unter Anfangs-Tag
- ...

Aber vielleicht ist das zu strikt. Wenn wir das wollen, brauchen wir auch einen Formatierer.

ergo-furrer, 2023-08-25:
> > Und was ist mit den Typen? Wollen wir die mit Grossbuchstaben beginnen und Nicht-Typen klein?
> 
> Auch OK. @ergo-furrer was meinst Du dazu?
Hmm, ich würde für komplexe Typen mit Gross- primitive Typen mit Kleinbuchstaben machen (in Analogie zu Java). Simple types würde ich mit Kleinbuchstaben beginnen.

ergo-furrer, 2023-08-25:
> ch kann aber auch mit der Definition von @ergo-furrer leben.

Simple type in Grossbuchstaben ist für mich auch OK. Dann sind alle selbstdefinierten Typen gross geschrieben.

**Zu Suggested Solution Simon**
Mit Vorschlag einvestanden.

**Zu Diskussionsgrundlage Arnd**
 Formatierung:
- keine Tabs OK
- Einrücken: 2
- Zeilen 120: OK, Ausnahmen erlaubt
- Nur immer ein <element>, <sequence>, ... in einer Zeile: Im XSD Ja. Im XML auch mehrere Elemente pro Zeile.
- Wenn ein Tag über mehrere Zeilen geht (z.B. Beschreibung) Ende-Tag auf eigene Zeile unter Anfangs-Tag: OK

ergozeller, 2023-08-25:
@ArndGerns Description aktualisiert, ready for implementation.

ArndGerns, 2023-08-28:
Only partially solved.

closed 2025-09-18 by mkrebs81