SmartGridready/SGrSpecifications #140: REQ - Modbus BitRank with Coils and Discrete Inputs

ergozeller, 2023-09-07:
## User Voice
The XSD currently allows for an optional bitRank on any modbus register. For holdingRegister and InputRegister this does not make sense. For coils and discreteInput regsiters we will run into trouble because modbus does not allow to directly access a single bit. the XSD should support what modbus actually can

### Context
Modbus provides the following functionality:
- register types `holdingRegister` and `inputRegister`: read and write is done in multiples of 16bit registers
- register types `coil` and `discreteInput`: read and write is done in multiple of bit, but always starting at the LSB. Modbus does not support to selectively read individual higher-ranked bits.

SGr data types
- SGr modbus data types `enum` and `bitmap` allow to selectively map individual bits to their respective states
- SGr modbus data type `boolean` will result in false for every coil/input off, and true if at least one coil/input is set

Currently no product XML uses coils or discreteInputs.

### Problem
The modbus data type element `<bitRank>` does not help to read out individual bits.
Furthermore, we currently have 341 usages of `<bitRank>`, all set to 0, and none non-zero. Some product XMLs even ommit `<bitRank>`. Therefore `<bitRank>` currently is not used at all.

### Suggested Solution

- Remove `<bitRank>` from XSD and product XMLs
- Single (and multiple) bits can be easily mapped to `enum` and  `bitmap`, no change needed
- We could convert modbus enums and bitmaps to generic boolean by defining a false as "firstEnum is set" or "no bitmap state is set". Alternatively we could extend the modbus boolean data type with a bitmask attribute.

## Acceptance Criteria
List the criterias that need to be met to declare this feature as implemented. This section must be complete and agreed upon before starting implementation.
- [ ] Solution must do this
- [ ] Application should behave like that

## Open Questions
List any open questions that need to be clarified before or during implementation.

**Additional context**
For enum and bitmap implementation see #111

ChrisBroenni, 2023-09-07:
_**>you wrote:** Modbus does not support to selectively read individual higher-ranked bits._

Coils and inputs use bit addressed formats.  
Check
03_Technische Spezifikation\11_Standardisierung_nur_interne_Verwendung\08_nicht_formal\Modbus
The use of "bitrank" is inherited from the XSD of IEC CD2 61850-80-5 TS : IEC 2020
11_kommission_technik\03_TechnischeSpezifikation\11_Standardisierung_nur_interne_Verwendung\07_CMTE\IEC\TC57\57_2250E_CD_61850-80-5_ModbusMapping.pdf
(its located in TC57 as it was only @ CD2 status in 2020)
The Schema is at 11_kommission_technik\zz_legacy\Interface **Factory\externeSchemaFiles**

**How To:**
To write Coil bit-level data, use Function Code 5 to write a single Coil (bit), or Function Code 15 to write multiple Coils (bits).
Be sure to use bit-level addresses to the module's internal database for any bit-level Modbus Commands. To find the bit level address for any bit in the module's internal database, use the formula: Register number times 16, plus the bit position number (plus 1 if you are telling a Modbus Master what address to poll when the MCM is a Slave.)

This assumes the first internal database register is Register Address 0 and the lowest order bit of any register is Bit 0.
So, for instance, if you want to write the value of Bit 10 of module database register address 100, the calculation would be:
100 * 16 = 1600 + 10 = 1610 as the Internal Address to use in the bit-level command if the MCM is the Modbus Master, or...
100 * 16 = 1600 + 10 + 1 = 1611, if the MCM is a Slave to a remote Modbus Master.
Bit address = (Word address * 16 + Bit rank (0 to 15))

The Packed Boolean formats behave as follows:

Packed Boolean: responds to read holding register function codes. Each input register (to the module) corresponds to one bit in the single output register of the module.

Packed Boolean for Coils: same as Packed Boolean except that the inputs are additionally mapped as coils in response to the Read Coil Status function code.

Packed Boolean for Inputs: same as Packed Boolean except that the inputs are additionally mapped as inputs in response to the Read Input Status function code.

### I recommend to close this issue.

ergozeller, 2023-09-07:
@ChrisBroenni OK thanks for the explanation, understood. 😃 

We will leave bitRank in the XSD as is (@ergo-furrer this is important for java implementation)

However I'll still remove the unneeded bitRanks on HoldingRegister and InputRegisters in the product XMLs

ChrisBroenni, 2023-09-07:
NO, its NEEDED

ergozeller, 2023-09-08:
- bitRank stays in the XSD (needed for coils and discreteInput)
- XMLs have been purged of unneeded bitRank=0 statements for HoldingRegisters and InputRegisters.

closed 2023-09-08 by ergozeller