SmartGridready/SGrSpecifications #157: Fix broken Hoval/CTA product declarations

ergozeller, 2023-09-19:
**Problem**
With #111 not all enums of hoval and cta product declarations have been merged correctly, and need rework

SGr_04_0017_xxxx_HOVAL_HeatPumpV0.2.1.xml
- hovHPOpModeCmd: der Enum hat neu 6 Werte. während er früher nur 4 Werte hatte. ist das OK?
- hovHeatCoolCtrlOpModeCmd: die Reihenfolge der Enums ist neu anders (manual cooling war 6, ist neu 7). ist das OK?
- hovHeatCoolOpState, hovHeatCoolCtrlOpModeCmd, hovHeatCoolOpState, hovHeatCoolCtrlOpModeCmd, hovHeatCoolOpState, 
-  DomHotWOpMode, hovDomHotWState, hovActBufferState: hier ist jeweils nur ein enum Wert angegeben. ist das OK?
SGReadyOpModeCmd, SGReadyState: hier hats noch einen alten iopEnumMapper drin. Ich habe das gefühl die Ordinals sind nicht 1234, sohern eher 255, 2, 0, 1, 3, oder sowas. ich verstehe den iopEnumMapper nicht...

SGr_04_0033_0000_CTA_HeatPumpV0.2.1.xml
- ctaDomHotWOpMode: mir scheint hier wären die falschen literals drin

ChrisBroenni, 2023-09-19:
Wie gesagt. Die WP’s haben noch nicht eingearbeitete Anpassungen welche wir nach dem Merge abarbeiten.

ergozeller, 2023-09-19:
@ChrisBroenni 
ne, bei der Einführung der neuen Enum Strukturen sind bei Hoval/CTA Dinge Infos verloren gegangen. Das korrigieren @ergo-furrer  und ich, so dass der neue Master wieder auf dem gleichen Stand ist wie der alte Master. Sonst haben wir später ein Durcheinander beim Merge deines Branches.

ChrisBroenni, 2023-09-20:
Eine Harmonisierung der WP enums im aktuellen Zeitpunkt ist  verlorene Zeit,  Hoval ist noch nicht nachgeführt und cta hat nach wie vor unbeantwortete Fragen. Gilt für alle WP enum "Investitionen" .
Wenn das parallel geändert wird gibt das IMHO viele offene Fragen "wer hat wann....."

ChrisBroenni, 2023-09-21:
IoP Enum mapper is a re-mapping of Smartgridready definitions into proprietary mappings.
This can be done in a easier way with the new enum methodology.

With CTA , some enums may disappear. There are different descriptions in different versions of documentation.  The simultor test is still not stable.

...28.Sep for to clarify

closed 2023-10-03 by ergo-furrer