SmartGridready/SGrSpecifications #190: REQ - XSD support for complex RestAPI requests

ginoagb, 2023-11-21:
During the implementation of the dynamic tariff functional profile #187 we noticed the impossibility of entering parameters in a restapi request.

For example
`https://api.tariffs.groupe-e.ch/v1/tariffs/vario_plus?start_timestamp=2023-11-17T00:00:00+01:00&end_timestamp=2023-11-18T00:00:00+01:00`

A solution to the problem could be the possibility of supporting:
- Query Params (Key, Value, Description)
- Headers (Key, Value, Description)
- eventually Body (?)

These parameter should be described in the XSD file.

The tariff array should be transformed in a standard format (similar to how we tranform a modbus register value)?

Additional issues:
- Reply is an array of objects that need to be interpreted. How can this be described in the EID.
- One of the elements of the object has a key which is the tariff_name. How can we support a flexible approch to keys?

## Acceptance Criteria
- [ ] The communicator should be able to use a getVal function to get the tariff array.

ginoagb, 2023-11-21:
@ArndGerns @ergo-furrer @DavidZoggFHNW

ArndGerns, 2023-11-21:
@DavidZoggFHNW, @ginoagb 

Ich habe das weiter mit Gino und Hans angeschaut und mache in den nächsten Tagen einen ersten Vorschlag. Den können wir dann zusammen gerne diskutieren.

Ich stimme Dir zu, dass ein Prduktname kein Key sein sollte. Vielleicht kann man das etwas anders strukturieren, so dass es zu einem Zeitraum eine "tariff_list" aus Paaren von Tarifname und Wert gibt,

DavidZoggFHNW, 2023-11-22:
Ja ich denke auch, dass eine Tarif-Liste die beste Lösung wäre.
Beste Grüsse
David