SmartGridready/SGrSpecifications #239: IP default address in EID must support IPV4, IPV6 and DHCP

ChrisBroenni, 2024-01-21:
## User Voice
_IP default address must support IPV4, IPV6 and DHCP_

### Suggested Solution

      <modbusInterfaceDescription>
        <modbusInterfaceSelection>TCPIP</modbusInterfaceSelection>
        <modbusTcp>
          <port>502</port>
          **<address>DHCP</address>**
          <slaveId>1</slaveId>
        </modbusTcp>
        <firstRegisterAddressIsOne>false</firstRegisterAddressIsOne>
        <bitOrder>BigEndian</bitOrder>
      </modbusInterfaceDescription>

### Alternatives (optional)
ideas are welcome

## Acceptance Criteria
Must be accpeted by
- Phyton group
- Java group
- Roundtable

**Additional context**
Was rised by HeatPump cleaning with manufactures

@ArndGerns 
@ginoagb 
@ergo-furrer

ginoagb, 2024-01-22:
@ergo-furrer the same should also apply to the REST API interface.

ArndGerns, 2024-01-24:
Hi @ChrisBroenni, @ginoagb 

how will the Communicator find the device? Just adding DHCP seems not to be enough to identify the device. We need something like a device name, a URL, or a mac address to find it.

Or how should that work?

Best regards,
Hans and Arnd.

ChrisBroenni, 2024-01-24:
uups, good point.. It must be device name as this goes with the device type, MAC address goes trhough instance.

ginoagb, 2024-01-30:
@ArndGerns I'm a little confused. Could you publish here the current content of the RestAPI.

In my expirience, having a default value for the IP of a device cause more issues than it solves. The person making the setup must know the IP address of the local device and configure it.

mkrebs81, 2024-07-09:
In my opinion, there should never be a default IP address or host name in any EID.
As @ginoagb mentioned, such addresses must be configured explicitly per device instance or in the communicator.

The only situation where I would consider setting a default address is a device EID which accesses a cloud service with a pre-defined hostname or URL.

ChrisBroenni, 2024-07-10:
This depends on the manufacturer: Many just specify "DNS" instead of a fix default address. The EID must describe the real image of the device "out of the box".  This is the reason why we defined it as a string.

ginoagb, 2024-07-11:
Since the standard is active DHCP, for those "rare" cases where a device is preconfigured with a static IP address, I would suggest writing it in the EID notes instead of creating an automation.

Also, for cloud IP addresses, these can change over time and, as @ArndGerns mentioned, it risks causing more confusion than anything else.

ChrisBroenni, 2024-07-11:
DHCP .... of course, not DNS :-O

ArndGerns, 2024-07-11:
How should a device should be found that comes with DHCP? IMHO the device will report some unique name or at least some MAC address.

So IMHO it should be OK to add the name in the ip address field - e.g. with `DHCP;who.am.i.com`.

If it comes with a MAC address, we could add `MAC` to the IP address field as a MAC in an EID does not make sense at all. Then it is up to the EMS to connect the device with its MAC to some instance.

And then we have the HOVAL where it can be configured on the device whether it is a static IP or DHCP. In that case, I have defined a configuration value `{{address}}` that needs to be handled during installation. This could also help in the other cases.

ArndGerns, 2024-07-11:
Currently, the ip address is defined as quadruple of integers. This is wrong anyway for IPV6. So I'ld propose to remove that restriction. Possibly, it would make more sense to add an additional field with the IP address type.

mkrebs81, 2024-07-11:
IMHO all devices using any kind of IP connection can have either

- an IPv4 address
- an IPv6 address
- a DNS host name

set as address in their EID.

If the IP address is dynamically assigned using DHCP / DHCPv6, there are a few approaches to discover the actual IP address:

1. Configure your router with a DHCP MAC reservation, so that the device always gets the same IP address, and configure the EID with the aforementioned IP address. This is probably the most generic and reliable approach.
2. Use ARP with MAC address to detect the IP address. This _may_ work, but only inside the local subnet, and not across routers!
3. Use device-specific discovery method, e.g. via UDP broadcast/multicast.
4. Use a standardized discovery method, such as _mDNS_. Support for this must already be present in the device!

I do not see other means to _reliably_ detect a device's IP address, unless the device has a built-in discovery method.

ChrisBroenni, 2024-07-11:
Yep, IMHO thats correct.  As example EEBUS would have special pocedures being part of the EEBUS Plug&Play  TransportService / configuration layer 6/7 twin setup. 
Should then be hosted in the EEBUS TransportService class / "LEGO Block".