SmartGridready/SGrSpecifications #249: Check new functional profile for Smart me

ergo-furrer, 2024-02-08:
**Done if**
- [ ] Version is v1.0.0
- [ ] Reviewed by all relevant stakeholders
- [ ] Product declarations concerned correcly use this XML
- [ ] FP tested by at least one device and customer

ergo-furrer, 2024-02-08:
@ginoagb I tested the functional profile for Smart-Me and found some issues with the current profile.

1. I had problems with the JMES path expression `Values[?Obis=='1-0:51.7.0*255'].Value | [0] * 1000`.  The multiplication factor `* 1000` is not a valid JMES path expression. It is not supported by the JMES-path library we use in Java and severe online JMES-path checkers on the internet. However we can avoid that and use the EI-XML `<unitConversionMultiplicator>1000</unitConversionMultiplicator>` tag to achieve the same.

2. I am not sure whether the multiplication factor 1000 is really correct. The numbers returned by the Smart-Me API already really high (196930.0, 2498190.0116666667, 30901360.0...). It appears to me that the factor should be 0.001.

3. The JMES path value filter-criteria (such as `1-0:51.7.0*255`) within the EI-XML do not match the values returned within the Smart-Me API response. 

Filter-expression in EI-XML
```
<query>Values[?Obis=='1-0:3.7.0*255'].Value | [0]</query> 			--> MISSING in response
 <query>Values[?Obis=='1-0:23.7.0*255'].Value | [0]</query>			--> MISSING in response
 <query>Values[?Obis=='1-0:43.7.0*255'].Value | [0] * `1000`</query>	--> MISSING in response
 <query>Values[?Obis=='1-0:63.7.0*255'].Value | [0] * `1000`</query>	--> MISSING in response
 <query>Values[?Obis=='1-0:1.8.0*255'].Value | [0] * `1000`</query>	        --> OK present in response / ActiveEnergyACtot
 <query>Values[?Obis=='1-0:1.7.0*255'].Value | [0] * `1000`</query>	        --> OK present in response / ActivePowerACtot
 <query>Values[?Obis=='1-0:21.7.0*255'].Value | [0]</query>			--> MISSING in response
 <query>Values[?Obis=='1-0:41.7.0*255'].Value | [0] * `1000`</query>	--> MISSING in response
 <query>Values[?Obis=='1-0:61.7.0*255'].Value | [0] * `1000`</query>        --> MISSING in response
 <query>Values[?Obis=='1-0:31.7.0*255'].Value | [0]</query>			--> MISSING in response
 <query>Values[?Obis=='1-0:51.7.0*255'].Value | [0] * `1000`</query>        --> MISSING in response
 <query>Values[?Obis=='1-0:71.7.0*255'].Value | [0] * `1000`</query>         --> MISSING in response
 <query>Values[?Obis=='1-0:11.7.0*255'].Value | [0] * `1000`</query>	--> OK present in response / CurrentACN
 <query>Values[?Obis=='1-0:14.7.0*255'].Value</query>				--> MISSING in response 
```

Smart-Me server response:
```
 [	
	"DeviceId": "08fffe1c-f3ae-4afe-bd70-bc0a73d7ac31",
    "Date": "2023-12-12T18:27:48.1015006Z",
    "Values": [
        {
            "Obis": "1-0:1.8.0*255",
            "Value": 196930.0
        },
        {
            "Obis": "1-0:1.7.0*255",
            "Value": 0.0
        },
        {
            "Obis": "1-0:32.7.0*255",
            "Value": 231.7
        },
        {
            "Obis": "1-0:11.7.0*255",
            "Value": 0.0
        },
        {
            "Obis": "1-0:33.7.0*255",
            "Value": 0.0
        },
        {
            "Obis": "0-0:96.9.0*255",
            "Value": 294.9
        },
        {
            "Obis": "99-0:10.0.0*255",
            "Value": 2498190.0116666667
        },
        {
            "Obis": "99-0:11.0.0*255",
            "Value": 30901360.0
        }
    ]
```

And therefore my introspective test results with 'null' or empty values for most of the data points:
```
        Reactive Power		ReactivePowerACtot		      null	
        Reactive Power		 ReactivePowerACL1		      null	
        Reactive Power		ReactivePowerACL2>		      null	
        Reactive Power		 ReactivePowerACL3		      null	
Active Energy Total Import		 ActiveEnergyACtot		  196930.0	
          Active Power		  ActivePowerACtot		       0.0	
          Active Power		   ActivePowerACL1		      null	
          Active Power		   ActivePowerACL2		      null	
          Active Power		   ActivePowerACL3		      null	
               Current		       CurrentACL1		      null	
               Current		       CurrentACL2		      null	
               Current		       CurrentACL3		      null	
               Current		        CurrentACN		       0.0	
         Net Frequency		         Frequency
```

ginoagb, 2024-02-09:
Hi Hans, thanks for the test.
1. The list of OBIS codes can be found here https://doc.smart-me.com/interfaces/api#h.betpktex7mrg
2. Yes, it's not times 1000 but divided 1000 to obtain the kW or kWh.
3. Smart-me ist certifiying the 3-phase meter, but to test the communication I was using the 1-phase meter. Because of that, there are a lot of missing datapoints

closed 2024-02-26 by ergo-furrer