SmartGridready/SGrSpecifications #271: REST API for SGr library

mkrebs81, 2024-04-08:
As discussed in the last SGr round-table meeting, I suggest adding a _REST API_ to search and download
product profiles from the SGr library.

This will help application developers and EMS manufacturers when they need to access the SGr library from
their own applications. For example, to generate a QR code for devices or configure a new SGr device in the EMS.

The current SGr library web application is not really suited to be queried from an external application,
because the backend calls seem to return JSON data combined with javascript.

## API Endpoints

The REST API should provide the endpoints described below.

### Search

The search endpoint returns a list of product profile results, similar to the web application.

_Filters:_

- release status
- manufacturers
- product names
- product categories
- interface types
- function profile categories
- sgr levels

_Request:_

POST a JSON object containing the search filters

_Response:_

a JSON array containing the results, which include the same fields as in the web application search results,
and - most important - the unique identifier of the product profile (XML file name).

### Download Product Profile

The profile endpoint returns the product profile as XML data.

_Request:_

The unique identifier (XML file name) as path or query parameter

_Response:_

The XML data

## Authentication

The REST API may require a simple API-key-based authentication.

This would allow entitled application developers to access the API, but not just anyone.

mkrebs81, 2024-09-24:
@ergo-furrer I remember that you presented a preview of an API documentation during one of the last roundtable meetings. Are there any updates on this topic?
I'm just asking because I'm collecting information for the students who work on the QR code project.
If you only have documentation or an API specification, they can still create a mock.

mkrebs81, 2025-01-13:
@ArndGerns @ergo-furrer I have done a quick test of the library REST API after getting an overview in today's meeting.
It works great overall and seems to have all the features I wished for.

So far I only found two small issues:
1. When I call `/prodx/{eidname}` (or `/prods`, `/fps` and `/fpx`) with a non-existing file name, I just get a status 200 with empty body, instead of an error status like 404.
2. The content type of the response is always `text/plain`, instead of `application/json` for declaration properties and `application/xml` for XML content.

mkrebs81, 2025-01-14:
Is there a way to retrieve the list of valid filter values for each filter type?
e.g. all release states, levels etc.

ArndGerns, 2025-01-14:
> @ArndGerns @ergo-furrer I have done a quick test of the library REST API after getting an overview in today's meeting. It works great overall and seems to have all the features I wished for.
> 
> So far I only found two small issues:
> 
>     1. When I call `/prodx/{eidname}` (or `/prods`, `/fps` and `/fpx`) with a non-existing file name, I just get a status 200 with empty body, instead of an error status like 404.
> 
>     2. The content type of the response is always `text/plain`, instead of `application/json` for declaration properties and `application/xml` for XML content. **EDIT: ** The content type is correct when I set the `Accept` header in the request.

Fixed

> Is there a way to retrieve the list of valid filter values for each filter type? e.g. all release states, levels etc.

Will try to add something - but will take some days as I'm very busy with other projects.

ArndGerns, 2025-01-14:
> @ArndGerns @ergo-furrer I have done a quick test of the library REST API after getting an overview in today's meeting. It works great overall and seems to have all the features I wished for.
>
> So far [...]

fixed

ArndGerns, 2025-01-14:
> Is there a way to retrieve the list of valid filter values for each filter type? e.g. all release states, levels etc.

Will try to add something - but will take some days as I'm very busy with other projects.

ArndGerns, 2025-01-15:
I have added commands "frf" and "prodf" to retrieve the filters with possible values.

mkrebs81, 2025-01-16:
Thank you. I tested it with Postman and it seems to work fine as far as I can tell.

I'm looking forward to integrating this in our QR code solution and test lab software.