SmartGridready/SGrSpecifications #277: Add parameter for serial port name to Modbus RTU interface schema

mkrebs81, 2024-04-30:
I have noticed that the interface schema of Modbus RTU devices contains all serial port connection options, except the actual serial port.

I understand that the reason might be the still hard-coded parameters of most product profiles (EIDs).

However, since we have the _configurationList_ now, I suggest adding the serial port parameter.
If the serial port is available, I can
- get all necessary parameters directly from the SGr EID
- implement cases where multiple serial ports are present

ergo-furrer, 2024-05-03:
This is a good point. If I understand correctly, you propose to add something like:
```xml
    <modbusInterfaceDescription>
        <modbusInterfaceSelection>RTU</modbusInterfaceSelection>
        <modbusRtu>
          <comPort>{{comPort}}</comPort>
          <slaveAddr>1</slaveAddr>
    ........
```
and then:

```xml
  <configurationList>
    <configurationListElement>
      <name>comPort</name>
      <dataType>
        <string/>
      </dataType>
      ........
```
We have to keep in mind, that with the current Java CommHandler implementation the communicator needs to instantiate the RTU communication driver separately and the COM port is not needed to be in the EI-XML:

```java
     GenDriverAPI4ModbusRTU mbRTU = new GenDriverAPI4ModbusRTU();
     mbRTU.initTrspService("COM3", 19200);
     var modbusDevice = new SGrModbusDevice(devDesc, mbRTU), null);
    ....
```
However, with the EI-XML change as suggested above we can change the CommHandler code to automatically load and configure the modbus driver (or in general the communication drivers) depending on the EI-XML content, which is a good thing from my point of view.

mkrebs81, 2024-05-03:
Thank you for the feedback. The XML draft is exactly what I had in mind.

However, I would not change the instantiation of the modbus driver, because putting it inside the comm-handler would not allow multiple devices on the same RS485 bus.

As far as I understand, I have to use the same `GenDriverAPI4ModbusRTU` object when trying to access multiple devices on the same bus.

I am currently working on an "extended" `SGrModbusDevice`, which uses a shared driver instance and calls `setUnitIdentifier()` on the transport before calling `setVal` or `getVal`.

ergo-furrer, 2024-05-03:
_As far as I understand, I have to use the same GenDriverAPI4ModbusRTU object when trying to access multiple devices on the same bus._
Indeed, I did not think of that. Yes we should keep it as is! 

_I am currently working on an "extended" SGrModbusDevice, which uses a shared driver instance and calls setUnitIdentifier() on the transport before calling setVal or getVal._
Thanks, this sounds good! Tell me if you have the related pull-request ready.

mkrebs81, 2024-05-15:
@ergo-furrer I just noticed that you updated the SGr specifications.
Is it correct that all Modbus interface parameters are now returned as strings instead of enums?

One more comment regarding the serial stop bits options:
AFAIK, the value 0 is not allowed. only 1, 1.5 and 2.

ergo-furrer, 2024-05-15:
> @ergo-furrer I just noticed that you updated the SGr specifications. Is it correct that all Modbus interface parameters are now returned as strings instead of enums?
> 
> One more comment regarding the serial stop bits options: AFAIK, the value 0 is not allowed. only 1, 1.5 and 2.

Yes indeed. This might be a problem, seem that we need to find another solution. I am considering to rollback the change, but currently have no good solution for the {{xyz}} expressions with are strings.

mkrebs81, 2024-05-15:
Well, I had to convert the enums anyway, because they were not compatible with the Modbus transport's enums.
I think the solution is ok for now...

ergo-furrer, 2024-05-15:
> One more comment regarding the serial stop bits options:
> AFAIK, the value 0 is not allowed. only 1, 1.5 and 2.

That's an error.

mkrebs81, 2024-05-15:
> > One more comment regarding the serial stop bits options:
> > AFAIK, the value 0 is not allowed. only 1, 1.5 and 2.
> 
> That's an error.
> 
> **_Update:_** Hmm, it validates on my side. The regex pattern contains 0 as well: "({{.+}})|0|1|1.5|2"

Sorry, I need to be more precise...
The value 0 is not allowed according to the serial protocol standard. At least one stop bit is always required.

ergo-furrer, 2024-05-15:
> Sorry, I need to be more precise...
> The value 0 is not allowed according to the serial protocol standard. At least one stop bit is always required.

OK, I see. I was not aware of that. The enum has a 0-value too. I think we should remove 0 on both.

ergo-furrer, 2024-05-16:
@mkrebs81 

> Well, I had to convert the enums anyway, because they were not compatible with the Modbus transport's enums.
> I think the solution is ok for now...

I had a discussion with Arnd, and we decided to bring up the issue reagarding returning Strings at the Roundtable coming Tuesday. So if you can work on this branch for now we would wait with the merge to master until coming Tuesday. Is that OK for you?

mkrebs81, 2024-05-16:
It's fine for me, I probably won't do further work on this project before Tuesday. I am currently using a static map to get the correct enum values from the string literals.

closed 2024-05-23 by ergo-furrer