SmartGridready/SGrSpecifications #328: REQ - Need for method to get REST data point query parameter placeholders?

mkrebs81, 2024-09-24:
## Situation

I have heard in the latest roundtable meeting that some EIDs with REST interface require dynamic query parameters, for example in dynamic tariff EIDs.

These parameters are specified via placeholders, similar to interface configuration parameters, and are replaced on every request.
While the configuration placeholders are specified in the `<configurationList>` structure, there is no equivalent for those dynamic REST query parameters.

## Inquiry

Do we need a parameter list inside the REST data point specification for such cases?

It might be helpful for testing and also parameter validation.

The schema would be very similar to the `<configurationListElement>`, therefore implementation would not be too difficult.

ergo-furrer, 2024-09-25:
I think i this is a good point and we should add it.
Yesterday I was thinking of that too, when you asked me about the parameters.

ginoagb, 2024-10-22:
@ergo-furrer hat das folgende Vorgeschlagen:

Betreffend Unterscheidung Konfiguration und Parametern sehe ich 2 Möglichkeiten, hab's aber noch nicht mit Arnd und Mathias besprochen:
Prefix (conf_xyz und par_xyz) oder Unterscheidung in der Klammerung ( {{ confVal}} [[parVal]] ).
 
Ich denke der Prefix hat den Vorteil, dass man schon am Namen sieht, worum es sich handelt. Die Unterschiedliche Klammerung bringt möglicherweise Vorteile in der Validierung des EI-XML und dass nicht versehentlich Konfigurationswerte als URL-Parameter verwendet werden.

ginoagb, 2024-10-22:
For me, it's important that a user immediately understands whether it's a **dynamic** or **config** parameter. 

I would use these two names to define the different types of parameters in the documentation or as prefixes.

This question remains: if a parameter is initialized at the start and then used dynamically, is it still _dynamic_, _config_, or _something else_?

mkrebs81, 2024-10-22:
Ich schlage folgende angepasste Variante vor:

1. die dynamischen Parameter in der `restApiDataPointConfiguration` deklarieren, analog zu configurationList (Name + Datentyp)
2. die Platzhalter **aller** request-spezifischen Parameter mit `[[...]]` annotieren, damit sie nicht beim Laden des EID ersetzt werden. Das gilt auch für alle Write-Requests wo `[[value]]` gesetzt wird.

mkrebs81, 2025-02-11:
@ArndGerns @ergo-furrer ich habe vorhin herausgefunden, dass der Typ _FunctionalProfileDataPoint_ (`FunctionalProfileFrame.xsd`) ein Element _requestParams_ vom Typ _RequestParam_ (`BaseTypes.xsd`) enthält.

Die DP-Spezifikation einer Produkt-EID hat dieses Element allerdings nicht.

Woher kommt dieser Unterschied?

Theoretisch könnte man diese Request-Params als dynamische Parameter verwenden, oder nicht?
Wobei der Parameterlist-Entwurf aus dem vorherigen Post so oder so besser aussieht.

mkrebs81, 2025-03-31:
> Mein Vorschlag zur Diskussion:
> 
>     * eine `<parameterList>`  vor dem `<dataType>` element.
> 
>     * Aufbau der Parameter Liste analog `<configurationList>`
> 
>     * Eckige Klammern [[....]] zur Unterscheidung und Vermeiden von Fehlern.
> 
>     * `parameterList`  ist nur für Interfaces definiert die auch Parameter untestützen. Aktuell ist das nur Http/REST.
> 
> 
>              <restApiDataPointConfiguration>
>                 <dataType>JSON_object</dataType>
>                 <parameterList>
>                   <parameterListElement>
>                     <name>start_timestamp</name>
>                     <dataType>JSON_string</dataType>
>                   </parameterListElement>
>                   <defaultValue>1970-01-01T00:00:00+00:00</defaultValue>
>                   <parameterDescription>
>                     <textElement>
>                       Begin of tariff periods to be queried. Format according ISO 8601: YYYY-MM-DDThh:mm:ss.sTZD.
>                     </textElement>
>                     <language>en</language>
>                     <label>Start timestamp</label>
>                   </parameterDescription>
>                   <parameterListElement>
>                     <name>start_timestamp</name>
>                     <dataType>JSON_string</dataType>
>                   </parameterListElement>
>                   <defaultValue>9999-01-01T00:00:00+00:00</defaultValue>
>                   <parameterDescription>
>                     <textElement>
>                       End of tariff periods to be queried. Format according ISO 8601: YYYY-MM-DDThh:mm:ss.sTZD.
>                     </textElement>
>                     <language>en</language>
>                     <label>Start timestamp</label>
>                   </parameterDescription>
>                 </parameterList>
>                 <restApiServiceCall>
>                   <requestHeader>
>                     <header>
>                       <headerName>Accept</headerName>
>                       <value>application/json</value>
>                     </header>
>                   </requestHeader>
>                   <requestMethod>GET</requestMethod>
>                   <requestPath>/v1/tariffs</requestPath>
>                   <requestQuery>
>                     <parameter>
>                       <name>start_timestamp</name>
>                       <value>[[start_timestamp]]</value>
>                     </parameter>
>                     <parameter>
>                       <name>end_timestamp</name>
>                       <value>[[end_timestamp]]</value>
>                     </parameter>
>                   </requestQuery>
>                   <responseQuery>
>                     <queryType>JMESPathMapping</queryType>
>                     <jmesPathMappings>
>                       <mapping><from>[*].start_timestamp</from><to>[*].start_timestamp</to></mapping>
>                       <mapping><from>[*].end_timestamp</from><to>[*].end_timestamp</to></mapping>
>                       <mapping><from>[*].vario_grid</from><to>[*].integrated[*].value</to></mapping>
>                       <mapping><from>[*].unit</from><to>[*].integrated[*].unit</to></mapping>
>                     </jmesPathMappings>
>                   </responseQuery>
>                 </restApiServiceCall>
>               </restApiDataPointConfiguration>
> 
>     * Für MQTT könnten getVal()-Parameter auch Sinn machen. Ist aber in SGrJava noch nicht implementiert.

Mein _finaler_ Vorschlag zur Umsetzung im Hinblick auf Demo-Projekte (z.B. dyn. Tarife):

1. Struktur wie vorgeschlagen
2. gleich für REST _und_ MQTT umsetzen
3. Umstellung auf eckige Klammern weglassen, da dies wahrscheinlich Probleme mit der Abwärtskompatibilität und existierenden EIDs verursacht. Es muss aber dokumentiert werden, dass die Parameter-Namen dann _nicht_ in der ConfigurationList auftauchen dürfen!

mkrebs81, 2025-04-07:
Nach dem Studium der aktuellen Spezifkation  ist mir aufgefallen, dass es noch eine Anpassung FP-Spezifikation braucht, damit diese konsistent mit den EIDs ist.

Der FP-Datentyp `requestParams` sollte _entfernt_ werden (wird aktuell sowieso nur beim dyn. Tarif benutzt) und durch eine FP-spezifische Variante der oben genannten ParameterList ersetzt werden.
Im Unterschied zum EID mit FP-spezifischen Datentypen und ohne Default-Wert.

mkrebs81, 2025-04-08:
@ergo-furrer I have created drafts of the final dynamic tariff FP and EIDs, which contain the not-yet-implemented parameter list and square brackets: https://github.com/SmartGridready/SGrSpecifications/tree/finalize-dynamic-tariffs

you can use this as reference when you modify the XSD specification.

mkrebs81, 2025-04-11:
@ergo-furrer thank you for implementing the dynamic parameters in the EID spec. I quickly tested it locally and it seems to work.

There are two more steps that are required on the specification side:

1. the same XSD structure must be applied to the functional profile XSD and replace `requestParams` there. just without default value.
2. we must replace `{{value}}` with `[[value]]` in all EIDs with writable data points

ergo-furrer, 2025-04-11:
> There are two more steps that are required on the specification side:
> 1. the same XSD structure must be applied to the functional profile XSD and replace `requestParams` there. just without default value.
> 2. we must replace `{{value}}` with `[[value]]` in all EIDs with writable data points

Indeed, i missed that. I will make the required changes. 

With the {{value}} to [[value]] change we must keep in mind, that EI-XML's will not work with CommHandler  versions <= V2.3.0 anymore.

mkrebs81, 2025-04-11:
I checked the whole repository, and it affects no published EI-XML.

And if necessary, a modified release 2.3.1 etc. can be easily made.

closed 2025-04-14 by mkrebs81