SmartGridready/SGrSpecifications #337: REQ - Introduce Field with Test Status?

mkrebs81, 2024-10-15:
## Current situation

EIDs have a field which shows the publication status.
However, there is nothing that helps tracking the status of EID tests.

## Suggestion

Suggested by @DavidZoggFHNW:

It would be useful to track which EIDs have already been tested, in addition to the published status.

mkrebs81, 2024-10-16:
according to meeting 16.10.

separate `testState`

mkrebs81, 2024-11-25:
Suggestions from @ArndGerns :

nach unserem Test-Lab-Meeting hier noch ein Diskussionsvorschlag für die Status zum Testen (jeweils von geringwertig zu hochwertig):
- None = ungetestet
- Tested = getestet durch "irgendwen"
- Vendor = getestet durch Hersteller incl. Testprotokoll
- Verified = getestet durch FNHW incl. Testprotokoll
oder nur
- None = ungetestet
- Tested = getestet durch Hersteller incl. Testprotokoll
- Verified = getestet durch FNHW incl. Testprotokoll
Was passiert, wenn z.B. Chris Brönnimann testet? Ist das dann auch Verified?

Und: Das muss sicher auch auf der Web-Seite dargestellt werden.
- Soll es dafür eine Spalte geben?
- Soll es dafür einen Filter geben?

Bitte kommentieren.

DavidZoggFHNW, 2024-11-25:
Ich wäre für die zweite Variante (None - Tested - Verified), weil einfacher. Unter "Tested" könnte auch irgendjemand sonst testen, nicht zwingend der Hersteller selbst.

DavidZoggFHNW, 2024-11-25:
Zur Website: Ich wäre dafür, dass auf library.smartgridready.ch bei den Produkten unmittelbar neben der Spalte "Releasestatus" eine neue Spalte "Teststatus" eingeführt wird.

mkrebs81, 2024-12-09:
Beschluss 09.12.2024:

- 4 Zustände --> None, Tested, Confirmed, Verified
- Filter in Library
- Spalte

closed 2024-12-18 by ArndGerns