SmartGridready/SGrSpecifications #341: REQ - Evaluate S2 Interface

mkrebs81, 2024-10-22:
Based on input from @ChrisBroenni on 22.10.24:

S2 may be required in the future:
- data formats
- not a communication interface like Modbus/REST/MQTT

See https://github.com/flexiblepower/s2-python for a Python implementation

TBD: what are the requirements for the SGr specifications?

ChrisBroenni, 2024-11-18:
[S2_In Short4Techies.pdf](https://github.com/user-attachments/files/17803774/S2_In.Short4Techies.pdf)
[S2-Whitepaper-202309-1.pdf](https://github.com/user-attachments/files/17803779/S2-Whitepaper-202309-1.pdf)
[20241019_S2_8.2_3_5.General.xlsx](https://github.com/user-attachments/files/17803797/20241019_S2_8.2_3_5.General.xlsx)
[20241019_S2_8.4.1_PowerEnvelopeBased_CTRL.xlsx](https://github.com/user-attachments/files/17803798/20241019_S2_8.4.1_PowerEnvelopeBased_CTRL.xlsx)
[20241019_S2_8.4.2_PowerProfileBasedCTRL.xlsx](https://github.com/user-attachments/files/17803799/20241019_S2_8.4.2_PowerProfileBasedCTRL.xlsx)
[20241019_S2_8.4.3_OperationModeBasedCTRL.xlsx](https://github.com/user-attachments/files/17803801/20241019_S2_8.4.3_OperationModeBasedCTRL.xlsx)
FillRate BasedCTRL and  DemandDriven BasedCTRL will follow.
contact ChrisB for to get EN50491-12-2 (!! SmartGridready members only for strictly non commercial use !!)

mkrebs81, 2024-12-09:
A proposal for a student project at FHNW has been published. I will update the comment as soon as it has been chosen (or not).

ChrisBroenni, 2025-01-13:
**Notes about discussion with KNX Technical Board, Joost Demarest**

- 	SGR confirmed that it would be involved in a student’s thesis work, whereby SGR is currently investigating which protocol to use for this: an open source implementation of TNO exists based on web sockets, but it is incomplete. 

- 	KNX confirmed that its KNX IoT (available as open-source stack from Gitlab) could be used between the CEM and the device’s resource manager. Either the DPTs could be used as suggested in the draft application specifications that were designed some years ago (and which could be provided by KNX) or these DPTs could be adjusted for IoT (as some of the DPTs in the draft application specifications considered that KNX classic useful data per telegram is limited). KNX offered to give support for the adjustment of the DPTs. 

- 	It was pointed out that the Point API open stack comes with a demo application that can be directly compiled for windows, of which the DPTs would then have to adjusted to those needed on the S2 interface. 

- 	KNX confirmed that it has plans to extend the current open-source stack also with P2P communication with TLS security, but does not have any plans for a solution based on websockets (which is a solution that is foreseen for the 3rd Party API (not targeted to embedded devices) but for which open source software is currently not available from KNX). 

- 	All needed tools (e.g. KNX Manufacturer Tool to create ETS product entries) could be made available in the framework of the **KNX scientific partnership, either for the university of the student or directly for SG**R (which could as a research body qualify for this type of partnership). 

[KNX IoT Open Source is available here](https://github.com/[KNX](https://github.com/KNX-IOT)-IOT)
The https://github.com/KNX-IOT/KNX-IOT-STACK is written in C language

The S2 Test implementation should  built on top of the available   [Example in CPP](https://github.com/KNX-IOT/KNX-IOT-Virtual)

However, the github information needs to verfied with ahaenel@knx.org  .

**NEXT STEPS**
We ( @ChrisBroenni , @mkrebs81 , @ArndGerns ) should sit together in order to evaluate the scope of the students work:

1.   Do we move on with a https://s2standard.org/ liaison  and inherit KNX outside of the Students work
     This could be done in Python as listed at the start of this issue
2. Do we take tha challenge and attack the KNX adventure as it may (or may not?) have a stratecial effect on our KNX liason
3. What is the requirment for to define the Students work fro the FHNW
4. Scale and define the content we provide towards the students

mkrebs81, 2025-01-13:
I suggest we discuss this topic further, as soon as we know if the student project was chosen (I should know it by the 20.01.2025).

mkrebs81, 2025-01-16:
Unfortunately the proposed student project was not chosen. :(

We had much more proposals than students in this semester, and perhaps the project's challenges seemed too difficult or not attractive enough.

Next steps to be discussed...

ChrisBroenni, 2025-04-08:
**Result of TNO meeting related to Open Source (related to "S2") coordination:**
TNO and Chris agreed that SGr is welcome publish and  bring to market MQTT OpenSource paired with the existing JSON / JSON-Schema  from TNO.
To bring it international market in NL could be brought  coordinated with https://flexible-energy.eu/ (potential liaison),  our Austrian network and last but not least with the grid connection point initiative of https://unit-e2.de/media/250131_cit_e_life_broschuere_i.pdfm,  D-A-CH-CZ group / VSE.

It can help establishing MQTT V5 https://www.hivemq.com/blog/mqtt5-essentials-part4-session-and-message-expiry/#heading-mqtt-5-s-message-expiry-interval paired with JSON / JSON-Schema in our international network AND make it broader.
This way, the value of SmartGridready becomes much more significant.

**Question:**
Can we give the ok to "unhold" S2 and apply  MQTT V5  on top of the available  TNO S2 material and start marketing it?

@mkrebs81 
@ergo-furrer 
@ginoagb 
@kevin-arm