SmartGridready/SGrSpecifications #359: Misanderstanding naming for  device configuration data  (configurationList)

ChrisBroenni, 2024-12-20:
**Describe the misunderstanding**

The item **_"configurationList"_**   has been confusing for EID editors / app developers as  it was not clear,  what element configurationList /  modbusInterfaceSelection  shows the default out of the box setting 

**Meaning**
configurationList :  is the DOCUMENTATION of parameters
modbusInterfaceSelection:  the REAL VALUES

**What do **
 We search a better naming for creating an intuitive  understanding
WHO IS THE DEER TO FIND IT ?
 
 ** Solution proposals**
1) add  configurationList as Element of deviceInformation
2) rename  configurationList  in configInfoList , configInfoListElemnt

Please add further solution proposals

mkrebs81, 2025-01-06:
I suggest improving the documentation, as renaming the element would break the specification and require modification of all EIDs.

mkrebs81, 2025-04-01:
will be added to documentation tasks