SmartGridready/SGrSpecifications #362: Stiebel-Eltron Internet Service Gateway ISG - Incorrect ENUMs?

mkrebs81, 2025-01-13:
**Describe the bug**

Using the EID `SGr_04_0015_xxxx_StiebelEltron_HeatPump_V1.0.0.xml` with a real controller delivers incorrect data point results.

- [ ] _SG-ReadyStates/SGReadyState_ returns invalid ordinal value `0`, resulting in `UNDEFINED`
- [ ] _HeatPumpBase/HPOpModeCmd_ cannot be set to `STI_WP_EMERG_OP` (ordinal `0`)

**To Reproduce**
Steps to reproduce the behavior:

_SG-ReadyStates/SGReadyState:_
1. Connect to device using commhandler library
2. read data point _SG-ReadyStates/SGReadyState_
3. See invalid value

_HeatPumpBase/HPOpModeCmd:_
1. Connect to device using commhandler library
2. set data point _HeatPumpBase/HPOpModeCmd_ to any other value than `STI_WP_EMERG_OP`
3. read data point again and verify that value is correct
4. set data point _HeatPumpBase/HPOpModeCmd_ to `STI_WP_EMERG_OP`
5. read data point again, value has no changed

**Expected behavior**
- _SG-ReadyStates/SGReadyState:_ a valid ENUM value should be returned
- _HeatPumpBase/HPOpModeCmd:_ not sure what the correct behaviour is, ask manufacturer

mkrebs81, 2025-01-14:
_SGReadyState_ is defined as:
- `HP_LOCKED` = 1
- `HP_NORMAL` = 2
- `HP_INTENSIFIED` = 3
- `HP_FORCED` = 4

Should it be 0...3 instead? since the device returns 0 from its Modbus register...

mkrebs81, 2025-01-14:
Changed the ENUM values in commit https://github.com/SmartGridready/SGrSpecifications/commit/e1ddd2c22e220284f38d1208ec8336bf0cb61e89

**Could somebody who knows the controller please verifiy**

DavidZoggFHNW, 2025-01-21:
In my opinion it makes sense that the HeatPumpBase/HPOpModeCmd cannot be set to STI_WP_EMERG_OP, since this is an emergency state, which may be set through the HP only -> We "soften" the test for this case. The other cases behave like they should.

For the SG-ReadyStates/SGReadyState ENUM: In my opinion this could be a typical 0/1 problem (first entry is 0 instead of 1). Thus HP_LOCKED would then be 0, HP_NORMAL 1 etc. In the test lab we can see a reaction of the controller to the command, but we have to verify again if it is in the right way.

We can clarify those things with STIEBEL ELTRON.

ChrisBroenni, 2025-05-10:
May the companion commhandler code being lost?

WP_EMERG_OP  could be set  , page 8 in  20230427_StiebelEtronWP_cand.docx ( at  the test documentation)

I hope this helps ?

ChrisBroenni, 2025-05-12:
>Actually, in case of this Stiebel Eltron device, 0..3 does not work, because combining the 2 16bit registers into a single value results in something like 0,1,256,257.
Misunderstanding !: The **commhandler software needs to translate the enum int the 2 Registers** representing the contacts.
Harmonization of ISO layer 5 / layer 7 interoperability often needs specification & source code to ensure compatibility.
That is one of the reasons why we startet this architecture. There is a tag (or was ?) a Layer7exception definition given the hint for applying this additional piece of source code.
For the time being I do have no time / no budget to grab this out again.....

mkrebs81, 2025-05-12:
The commhandler already maps the enum's numerical value to the 2 registers, provided that
a) the numerical values are correctly defined in the EID, so they match the register values
b) the register addresses are consecutive

chrohrbach, 2026-11-05:
Thanks for the report and for #364, which fixed several real problems in this EID (the input-register type of `SGReadyState`, the redundant input data points, the 2m/4m synchronisation).

I compared the `SG-ReadyStates` data points with Stiebel Eltron's ISG Modbus documentation (blocks 5 and 6, chapter 8.2) and I think two sets of ordinals still need another look:

- Register 5001 is documented as 1 to 4 (1 locked, 2 normal, 3 intensified, 4 forced). The EID declares 0 to 3 (normal 0, intensified 1, locked 2, forced 3). The `0` seen here may have another cause: `SGReadyState` was still a holding register when the ordinals were changed in January 2025, and became an input register only in the last commit of #364. I cannot tell without a trace of the requests.
- 4002 is SG READY INPUT 1 (documented state 3) and 4003 is INPUT 2 (documented state 1). With a big-endian `int32U` across both registers, `HP_LOCKED` = 65536 sets input 1 and `HP_INTENSIFIED` = 1 sets input 2, which is the reverse of the documentation. I checked this with the Python CommHandler against a local stub that only logs requests, not on a device.

Proposed change: #452. @mkrebs81, since you have a controller at hand: could you check the raw values of 5001 and 4002/4003 on it (details in the PR)? A confirmation from Stiebel Eltron via the test lab (@DavidZoggFHNW) would settle it.