SmartGridready/SGrSpecifications #373: Validation Tool

mkrebs81, 2025-02-25:
Shall we integrate the FP / EID validation tool into this repository?

- language-specific code should not go into specification repository, only XML/XSD
- the tool is written in Java but not directly related to the SGr Java libraries

--> so I believe it is OK...

I suggest putting the tool in the following directory structure:

```
repo root
|_ tools
   |_ validator
```

mkrebs81, 2025-02-25:
@ArndGerns I have checked in a potential solution in the attached branch. Feel free to test it.

ArndGerns, 2025-03-31:
> [@ArndGerns](https://github.com/ArndGerns) shall we merge this into the master branch? I do not see any reason why not. you perhaps?

Yes, Please create a pull request.
(If I´d do it, there could be a deadlock with approvals.)

closed 2025-03-31 by mkrebs81